driftregion/iso14229 #4: Handling RX Data

imecar-github, 2022-11-17:
Hello. Thanks for the library.

I have a question about handling responses. Im currently sending uds data and it looks OK. But, How can I handle the response? I mean, 
How can I use the mockClientCANRxPoll function? 

I modified mockClientSendCAN with HAL library for sending data. But I can not manage mockClientCANRxPoll. 

Sorry for my bad English. Have a good day.

driftregion, 2022-11-17:
It looks like you're using ST's HAL: https://www.disca.upv.es/aperles/arm_cortex_m3/llibre/st/STM32F439xx_User_Manual/group__can__exported__functions__group3.html#gaf25698a35af7f78d01b036fcb80d81f3 

`HAL_CAN_GetRxMessage` returns `HAL_OK` if there was a message otherwise `HAL_ERROR`.

Try this:
```c
static enum Iso14229CANRxStatus mockClientCANRxPoll(uint32_t *arb_id, uint8_t *data, uint8_t *dlc) {
    if (HAL_OK == HAL_CAN_GetRxMessage(&hcan1, CAN_RX_FIFO0, &rxHeader, TxData))  {
	*arb_id = rxHeader.ExtId;
	*dlc = rxHeader.DLC;
	memmove(data,TxData,*dlc);
        return kCANRxSome;
    }
    else {
        return kCANRxNone;    
    }
}
```

imecar-github, 2022-11-17:
> It looks like you're using ST's HAL: https://www.disca.upv.es/aperles/arm_cortex_m3/llibre/st/STM32F439xx_User_Manual/group__can__exported__functions__group3.html#gaf25698a35af7f78d01b036fcb80d81f3 [...]

Yes I am using ST's HAL. Thanks for response. 

But whenever I want to debug code, mockClientCANRxPoll function is not triggered. 

```
Iso14229Client udsClient;
    	struct Iso14229ClientConfig cfg = {
    	    .phys_send_id = 0x18DA66AA,
    	    .func_send_id = 0x18da55aa,
    	    .recv_id = 0x18DAAA66,
    	    .p2_ms = CLIENT_DEFAULT_P2_MS,
    	    .p2_star_ms = CLIENT_DEFAULT_P2_STAR_MS,
  	        .link = &g.clientLink,
  	        .link_receive_buffer = g.clientLinkRxBuf,
  	        .link_recv_buf_size = sizeof(g.clientLinkRxBuf),
  	        .link_send_buffer = g.clientLinkTxBuf,
  	        .link_send_buf_size = sizeof(g.clientLinkTxBuf),
  	        .userCANTransmit = mockClientSendCAN,
  	        .userGetms = mockUserGetms,
  	        .userCANRxPoll = mockClientCANRxPoll,
  	        .userDebug = mockClientLinkDbg,
    	    };
```
Here is my configuration.

driftregion, 2022-11-18:
You can try either:

- disable the interrupt that calls HAL_CAN_RxFifo0MsgPendingCallback (perhaps the opposite of this issue: https://community.st.com/s/question/0D50X0000AvfKJTSQ2/solved-can-callback-never-call). This should allow you to use `HAL_CAN_GetRxMessage`.
- use a queue. Write messages to it in `HAL_CAN_RxFifo0MsgPendingCallback` and read from it elsewhere in your program.

imecar-github, 2022-11-19:
When I disable Interrupts, TX and RX functions triggered. Thanks.

closed 2022-11-19 by imecar-github