driftregion/iso14229 #7: TransferData Question

imecar-github, 2022-11-23:
Hi. I'm using SD CARD with FATFS. But TransferDataStream is using iostream. I could not use TransferDataStream with FATFS. 

My first question is, Is there any other else TransferData function with blocksize 128?
Second question is, how can I use TransferDataStream with FATFS ?

imecar-github, 2022-12-06:
`` if(fileEOF == false)
        	{
			if(file_counter > fileFileInformation.fileSize)
				{
					fileEOF = true;
					 return kISO14229_CLIENT_CALLBACK_DONE;
				}
        		file_willRet = 128;
        		if((fileFileInformation.fileSize - file_counter) < 128 )
        		{
        			file_willRet = fileFileInformation.fileSize-file_counter;
        			lastBlock = true;
        		}
        		   TransferData(client, file_blockSizeCounter, 130, fileFileInformation.buffer + file_counter, file_willRet);
        		   if(!lastBlock)
        		   {
        			   if(20 == file_canMessageCounter)
						{
							file_blockSizeCounter = file_blockSizeCounter + 1;
							if(file_blockSizeCounter == 256)
							{
								file_blockSizeCounter = 0;
							}
							file_counter+= 128;
							file_canMessageCounter = 0;
							g.serverRecvQueueIdx = 0;

						}

        		   }
        		   else
        		   {
        			   if((((fileFileInformation.fileSize - file_counter) / 7) + 1) == file_canMessageCounter)
						{
							file_blockSizeCounter = file_blockSizeCounter + 1;
							if(file_blockSizeCounter == 256)
							{
								file_blockSizeCounter = 0;
							}
							file_counter+= (fileFileInformation.fileSize - file_counter);
							file_canMessageCounter = 0;
							g.serverRecvQueueIdx = 0;
							fileEOF= true;
						}

        		   }
        		    file_canMessageCounter++;
        	        HAL_Delay(5);
        	        return kISO14229_CLIENT_CALLBACK_PENDING;
        	}
        	else
        	{
        		 return kISO14229_CLIENT_CALLBACK_DONE;
        	}

            return kISO14229_CLIENT_CALLBACK_DONE;`

I write something like this, but It does not work every file. Could you provide me an algorithm ?`

I write something like this, but It does not work every file. Could you provide me an algorithm ?

imecar-github, 2022-12-06:
Edit: 

`if (file_driverEOF == false) {
		if (file_counter > fileFileInformation.fileSize) {
			file_driverEOF = true;

		}
		file_willRet = 128;
		if ((fileFileInformation.fileSize - file_counter) < 128) {
			file_willRet = fileFileInformation.fileSize - file_counter;
		}

		TransferData(client, file_blockSizeCounter & 0xFF,
				file_willRet + 2,
				fileFileInformation.buffer + file_counter,
				file_willRet);
		if(client->state == kRequestStateSent)
		{
			 file_counter += 128;
				file_canMessageCounter = 0;
				file_blockSizeCounter++;
				g.serverRecvQueueIdx = 0;
		}

		HAL_Delay(5);
		return kISO14229_CLIENT_CALLBACK_PENDING;
	}

	return kISO14229_CLIENT_CALLBACK_DONE;
`

I just tried this code and everything seems fine!

driftregion, 2022-12-11:
> I just tried this code and everything seems fine!

OK. I'm closing this issue.

closed 2022-12-11 by driftregion