driftregion/iso14229 #11: How to use with PCAN

lushee, 2023-02-14:
Hi sir iam having a bit hard time figuring how to make this 14299.c , 14299.h to work with PCAN , 
1)I have pcan installed in my system , but don't know how to make it work with  code 14299.c since it does not have main function .  2)My main Aim is to check the data send from PCAN with out code for the respective diag services .
3)A little help on how to do it will be appreciated .

Regards.

driftregion, 2023-03-13:
Hi @lushee, 

iso14229 doesn't support PCAN at the moment.  However, adding support should be straightforward.

1. download the PCAN-Basic API. https://www.peak-system.com/PCAN-Basic.239.0.html?&L=1
2. use the PCAN-Basic  API functions (`TPCANStatus __stdcall CAN_Write(TPCANHandle Channel, TPCANMsg* MessageBuffer);`) to implment initialization, deinitialization, send, and receive functions.  see [examples/isotp-c_on_socketcan.c](https://github.com/driftregion/iso14229/blob/0.6.0/examples/isotp-c_on_socketcan.c) for a porting example.
3. compile with `-DUDS_ARCH=UDS_ARCH_WINDOWS` `-DUDS_TP=UDS_TP_ISOTP_C`

lushee, 2023-04-04:
Thank you I also have some doubts about DID's handling and where it is done can you please guide me  sir?
Regards.

lushee, 2023-04-04:
The Implementation and the modification are posted by me here please take a look, I am not very proficient please bear if my questions are amateur.
https://stackoverflow.com/questions/75916133/uds-services-implementation-in-c?noredirect=1#comment133910619_75916133

closed 2023-04-17 by lushee