driftregion/iso14229 #16: working with flexcan in s32ds

lushee, 2023-08-08:
Hi @driftregion iam facing an issue when working with the flexcan to uds currently this is how my initialization looks please guide me how to approach it :
1)Should i link the FLEXCAN to isotp-c to uds ?
2)i have made my custom flow control Should i directly link my flexcan to UDS ? [current approach]
i have been developing the code portable uds since last 6 months if possible can we connect and discuss ? please send me a mail to lm921570.ttl@tatamotors.com so i can setup a teams meeting.

driftregion, 2023-08-09:
Hi @lushee. I'm assuming that the FLEXCAN in your question is NXP's CAN driver "FlexCAN" and not this  https://paws.kettering.edu/~jpimente/flexcan/FlexCAN-architecture.pdf   Is my assumption correct?

lushee, 2023-08-09:
yes @driftregion  my FLEXCAN is of NXP you can find the same api's here although but my board is a different one
https://blog.csdn.net/weixin_44712171/article/details/124953964

driftregion, 2023-08-09:
OK. I understand that you're using NXP's FlexCAN driver. Could you please state the problem you're having with iso14229?

lushee, 2023-08-10:
the problem iam facing is that iam not able to fully utilise the code of iso14229 since iam not able to link the flexcan api handles properly to iso14229 currently this is my approach:

1)Connect PCAN usb  to can gpio pins,
2)send the request of uds in pcan gui,
3)the data is then recived in FlexCAN_Ip_Receive(INST_FLEXCAN_0, 2, &rxData, false);
4)the data is then passed to my buggy flow control code ,
4)then iam intializing UDSServer_t server; and copying data from flow control directly  to uds server buffer
status = FlexCAN_Ip_Receive(INST_FLEXCAN_0, 2, &rxData, false);
				if (status == FLEXCAN_STATUS_SUCCESS) {
				 uds_multipacket_frame_handler();
				 if(Final_Frame_Flag ==  1) {
					   memcpy( server.recv_buf, uds_buffer+1, uds_buffer_index);
					   server.recv_size = uds_buffer[0];
					   server.send_buf_size = UDS_BUFSIZE;
					   server.fn = evaluateServiceResponse;

					      // Calling the function through the function pointer
					      uint8_t result = server.fn(&server, kTpAddrTypeFunctional);
					  // server.fn = evaluateServiceResponse(&server,kTpAddrTypeFunctional);
					   uds_multipacket_frame_send_handler(server.send_buf);

				   }
                }
            }

driftregion, 2023-12-15:
Hi @lushee , I've uploaded example code for S32K144 here: https://github.com/driftregion/iso14229/tree/0.7.0/examples/s32k144

Let me know if this helps

driftregion, 2024-02-26:
Closing due to inactivity

closed 2024-02-26 by driftregion