driftregion/iso14229 #17: UDS client responding with a functional flow control frame to a physically addressed request

muehlke, 2023-10-04:
Hi @driftregion , 

when sending a physically addressed RDBI request to an ECU, that answers with a multi-frame RDBI response, the client sends two flow control frames, one that is physically addressed and the other functionally. Here is a log with `candump`:

```
(1688347721.008952) can0 18DAF140#103562FD13584355  /* response to RDBI request */
(1688347721.009126) can0 18DB00F1#301003  /* functionally addressed response even if the request was physically addressed */
(1688347721.009231) can0 18DA40F1#301003 /* physically addressed response */
(1688347721.012025) can0 18DAF140#215F544848000000
```

Sending two flow control frames makes the ECU answer twice, which is unnecessary and unwanted.

I'm not able to explain myself why this happens since in the function `tp_recv` the code for reading from the socket reads from the physical link first:

```
struct {
        int fd;
        UDSTpAddr_t ta_type;
    } arr[] = {{impl->phys_fd, kTpAddrTypePhysical}, {impl->func_fd, kTpAddrTypeFunctional}};
    for (size_t i = 0; i < sizeof(arr) / sizeof(arr[0]); i++) {
        ret = read(arr[i].fd, buf, count);
        ...
        ...
```
After reading from the physical link, the socket should get the answer it needs and then not read another time from the functional link. Do you have an idea what this could be happening? If you need more info to the context or execute code, just let me know and thanks in advance!

driftregion, 2023-10-12:
Hi @muehlke , I agree this is undesirable behavior. 

which transport layer implementation are you using?
 
if isotp-c, can you share the section of code where you call `isotp_on_can_message`?

muehlke, 2023-10-12:
Hi @driftregion , I'm using the Linux ISOTP kernel module.

driftregion, 2023-10-24:
We may be using the kernel module incorrectly. There's a flag `CAN_ISOTP_LISTEN_MODE` that probably needs setting on the socket that's used to send functional requests by the client:
https://elixir.bootlin.com/linux/v6.5.8/source/include/uapi/linux/can/isotp.h#L138

Could you try setting this flag and see if this solves the problem?

muehlke, 2023-10-25:
Hi,

I tried setting the flag as you mentioned like the following before setting the can_isotp_fc_options:

```
struct can_isotp_options opts;
memset(&opts, 0, sizeof(opts));
opts.flags |= CAN_ISOTP_LISTEN_MODE;
if (setsockopt(fd, SOL_CAN_ISOTP, CAN_ISOTP_OPTS, &opts, sizeof(opts)) < 0) {
    perror("setsockopt (isotp_options):");
    return -1;
}
```

and it solved the problem 👍  The only issue would be the method definition. It is necessary to distinguish between opening a socket for the physical and functional ID of the UDS client. Obviously if both sockets are opened with this set flag, then the client doesn't send a FC frame from its physical ID causing a timeout.  

One idea would be to change the definition to something like:

`static int LinuxSockBind(const char *if_name, uint32_t rxid, uint32_t txid, bool funcSocket)`

where ``funcSocket` is used to indicate whether the socket to be bound is a functional one:

```
if (funcSocket) {
    struct can_isotp_options opts;
    memset(&opts, 0, sizeof(opts));
    opts.flags |= CAN_ISOTP_LISTEN_MODE;
    if (setsockopt(fd, SOL_CAN_ISOTP, CAN_ISOTP_OPTS, &opts, sizeof(opts)) < 0) {
        perror("setsockopt (isotp_options):");
        return -1;
    }
}
```

to then change `LinuxSockTpOpen` to:

```
...
hdl->poll = tp_poll;
impl->phys_fd = LinuxSockBind(if_name, phys_rxid, phys_txid, false);
impl->func_fd = LinuxSockBind(if_name, func_rxid, func_txid, true);
...
```

driftregion, 2023-12-15:
This has been fixed in #21

muehlke, 2023-12-20:
I see, you moved the implementation to src/tp/isotp_sock.c

Thanks for addressing this issue!

closed 2023-12-20 by muehlke