driftregion/iso14229 #18: Compiler warnings after upgrading to release v0.6.0

muehlke, 2023-10-12:
Hello @driftregion again,

after upgrading to the new release v0.6.0 and compiling with the flags `-Wall` and `-Werror` I get these two errors:

1. [-Werror=int-to-pointer-cast]
```
 iso14229/iso14229.c:517:22: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
 517 |     *memoryAddress = (void *)tmp;
```
Regarding this error, does the `tmp` variable has to be of the type `long long unsigned int`? When I remove one `long` to make it `long unsigned int`, it compiles with no warnings (errors) - when using the `-Werror` flag warnings are made to errors. I would highly recommend to keep the standard of the library to compile with no warnings, no matter how "trivial".

2. [-Werror=stringop-truncation]
```
armv7vet2hf-neon-poky-linux-gnueabi/usr/include/bits/string_fortified.h:106:10: error: ‘__builtin_strncpy’ specified bound 16 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
```

First, I appreciate changing to the more secure variant `strncpy()`. I could fix this warning by changing the line 239 of `iso14229.c` from:

```
strncpy(ifr.ifr_name, if_name, sizeof(ifr.ifr_name));
```
to
```
strncpy(ifr.ifr_name, if_name, sizeof(ifr.ifr_name) - 1);
```

Let me know what you think about these warnings and whether you think these changes are worth making.

muehlke, 2023-10-25:
You addressed the second warning ([-Werror=stringop-truncation]) in #20 with commit `fb401496`, while the first warning ([-Werror=int-to-pointer-cast]) unfortunately remained unsolved. Would it be fine to change `tmp` from a `long long unsigned int` to a ``long unsigned int`? Or are you working on a 64 bit system?

driftregion, 2023-12-15:
this is also fixed in #21

muehlke, 2023-12-20:
I see the change in line 181 from src/server.c, you changed it from `long long unsigned int` to `uintptr_t`

Thanks for addressing this issue! The project is growing, kudos for the hard work!

closed 2023-12-20 by muehlke