driftregion/iso14229 #24: working with cannelloni as interface

pompushko, 2024-01-10:
Hello. 

Is it possible to add https://github.com/mguentner/cannelloni/blob/master/doc/tcp_protocol.md as can-bus interface?
I want to use your library on device(MCU) where is only TCP/IP connectivity.

Thank you

driftregion, 2024-01-10:
I don't see why not. You'll still need something to link the CAN frames
sent by via cannelloni/TCP together into messages to pass to the UDS layer.
Have a look at the isotp-c examples on the 0.7.0 branch to see some custom
transport implementations.

Have you considered DoIP? There already exists a python-udsoncan compatible
DoIP implementation. Some reference C implementations can also be found on
GitHub.

pompushko, 2024-01-10:
Can I donate for such feature :D?

driftregion, 2024-01-11:
Certainly. You can donate via https://github.com/sponsors/driftregion .

driftregion, 2024-01-13:
For support inquiries please contact me at iso14229dev@gmail.com

driftregion, 2024-02-26:
Closing due to inactivity

closed 2024-02-26 by driftregion