driftregion/iso14229 #26: isotp-c integration (updates)

SimonCahill, 2024-02-23:
Hi there!

A while ago you integrated my fork of the isotp-c library in this project.
I just wanted to inform you that several updates have been published since, and I thought it would be good to inform you.

The latest version is [v1.1.1](https://github.com/SimonCahill/isotp-c/releases/tag/v1.1.1).

driftregion, 2024-02-26:
Thanks Simon for this notification. I appreciate your work.
I'm still running a fork with a modified  `isotp_user_send_can` that accepts an additional `void *` so that multiple CAN interfaces may be used by one binary. Is this something you'd consider upstreaming?

SimonCahill, 2024-02-26:
Sounds like a good feature!

To maintain backward-compat. could you add this as an option in the Makefile and CMakeLists?
That way this feature could be enabled at compile-time.

Shoot me a PR and I'll see it gets merged :)

driftregion, 2024-02-26:
Sounds good. Next time I touch isotp-c I'll send a PR. It might be a while though. The library never has any problems. ;)

SimonCahill, 2024-02-26:
No worries, I'm looking forward to the contribution!
Take your time, it's no rush

driftregion, 2024-08-15:
Hi Simon, I took a stab at this here:

https://github.com/SimonCahill/isotp-c/pull/36

closed 2024-08-16 by SimonCahill