driftregion/iso14229 #37: UDS Client

guru0808, 2024-07-23:
Hi
I cannot find any example related to UDS Client usage. However i have figured out a way to read the response data and would like to verify if there is any other way. In simple terms I wait for idle state after executing request then read data from buffer
```
 case JOB_FUNC_W8_RESP:
            // IVEC_BSW_LOG(LOG_STRING, "Waiting for resp %d\n", l_i32Error);
            if(UDSClientPoll(&__gprv_UdsClient)==UDS_CLIENT_IDLE)
            {
                IVEC_BSW_LOG(LOG_STRING, "Request Complete.\n");
                **for (int i = 0; i < __gprv_UdsClient.recv_size; i++) 
                    printf("%02X ", __gprv_UdsClient.recv_buf[i]); // my data
                printf("\n");**//
                 __lprv_i32JobStepNo = JOB_FUNC_FINISH;
            }
        
        break;
```

driftregion, 2024-07-24:
Looks about right. There is an example here: https://github.com/driftregion/iso14229/blob/main/examples/linux_server_0x27/client.c

guru0808, 2024-07-24:
Right.
I also faced a scenario where i am requesting to change session from client to server. And i am receiving data bytes as following
`[0x7f,0x10,0x11]` which is  right as per my implementation. In such case, Am i supposed to parse response from recv_buff data because i'm not getting any failure notification. As per this block, my response is probably getting into the else block. Belowis a snippet from function _ClientHandleResponse.
```
 if (0x7F == client->recv_buf[0]) {
        if (kRequestCorrectlyReceived_ResponsePending == client->recv_buf[2]) {
            UDS_DBG_PRINT("got RCRRP, setting p2 timer\n");
            client->p2_timer = UDSMillis() + client->p2_star_ms;
            memset(client->recv_buf, 0, client->recv_buf_size);
            client->recv_size = 0;
            UDSTpAckRecv(client->tp);
            changeState(client, kRequestStateAwaitResponse);
            return;
        } else {
            ;
        }
    } 
```
Also, i don't see the inokation of `UDSClient.fn` .

driftregion, 2024-07-24:
You can use the `UDS_NEG_RESP_IS_ERR` option to enable errors in response to server NRCs. See:

https://github.com/driftregion/iso14229/blob/24c60fdab402765eb8ef6cabad111534827cbc9d/test/test_client_0x11_ECU_reset.c#L31C19-L31C38

> Also, i don't see the inokation of UDSClient.fn .

yes, that API is unfinished. Sorry about that.

momet83, 2024-10-11:
is it possible to upload the *.sln?

closed 2025-03-09 by driftregion