driftregion/iso14229 #40: Service mode receiving message generates bug (msg like "0x7DF 3E 80 ")

matlam1989, 2024-08-18:
In server mode , mcu just receive "0x7DF 3E 80 "，and debug msg always got like:
_\tp\isotp_c.c 124)ack recv
\tp\isotp_c.c 31)The link is full. Copying 2 bytes
\tp\isotp_c.c 69)just got 2 bytes on func link 
\server.c 901)len: 0_
Even,I unplug the CAN analyzer,and The above debugging information has not stopped.

driftregion, 2024-08-18:
Hi matlam1989, thanks.

There aren't enough details in your issue for me to understand what the problem is.

Can you please show an example program, OR describe in more detail what you expected to happen and what actually happened?

matlam1989, 2024-08-18:
I add "r->send_len"  check in server.c, but "r->recv_len "never initi to zero ,even client not send same msg
[image]
[image]

matlam1989, 2024-08-18:
I use the src file in the library ， no using the generated iso14229.c
[image]
[image]

driftregion, 2024-08-18:
Please post a complete example code. Thank you.

matlam1989, 2024-08-18:
I change :
[image]
[image]

matlam1989, 2024-08-18:
I use iar for arm 9.2,and define ISOTP_BYTE_ORDER_LITTLE_ENDIAN in the ide.

matlam1989, 2024-08-18:
more detail:
_const UDSISOTpCConfig_t tp_cfg =
{
    .source_addr=0x7C0,         
    .target_addr=0x7C8,          
    .source_addr_func=0x7DF,        
    .target_addr_func=UDS_TP_NOOP_ADDR,
    .isotp_user_send_can=send_can,
    .isotp_user_get_ms=millis,
    .isotp_user_debug=NULL,
    .user_data=NULL,
};

void init()
{
    assert(UDS_OK == UDSServerInit(&srv));
    assert(UDS_OK == UDSISOTpCInit(&tp, &tp_cfg));

    srv.tp = &tp.hdl;
    srv.fn = fn;
}

void loop()
{
    UDSServerPoll(&srv);
}_

driftregion, 2024-08-18:
There is not enough information for me see where the problem is.

Can you show the implementation of `isotp_user_send_can` and show the return value of this function?
Can you also share a log of CAN messages sent to and from the server?

matlam1989, 2024-08-18:
Sorry, I didn't see the message. 
Now , I debug again,and find Program call relationship：
  ssize_t UDSTpSend(struct UDSTpHandle *hdl, const uint8_t *buf, ssize_t len, UDSSDU_t *info) ; 
     ->static ssize_t tp_send(UDSTpHandle_t *hdl, uint8_t *buf, size_t len, UDSSDU_t *info)
       ->int isotp_send(IsoTpLink *link, const uint8_t payload[], uint16_t size)
         ->int isotp_send_with_id(IsoTpLink *link, uint32_t id, const uint8_t payload[], uint16_t size)
           ->static int isotp_send_single_frame(IsoTpLink* link, uint32_t id)
             ->final to my own define sendAPI 
               -> return ret = 8; //msg out ： 0x7C8  0 0 0 0 0 0 0 0
I can't give more can message in my company .Only partial UDS related messages.
[image]

driftregion, 2024-08-18:
Please share a [Minimum reproducible example](https://en.wikipedia.org/wiki/Minimal_reproducible_example).

matlam1989, 2024-08-19:
I use C++ for development, and there are very few changes to the library files.more detail:
#include "elog.h"
extern "C"{
#include "server.h"
#include "isotp_c.h"
}

namespace udsCfg{

extern "C" uint32_t millis();

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud);

UDSServer_t srv;
UDSISOTpC_t tp;

const UDSISOTpCConfig_t tp_cfg =
{
    .source_addr=0x7C0,             
    .target_addr=0x7C8,            
    .source_addr_func=0x7DF,        
    .target_addr_func=UDS_TP_NOOP_ADDR,
    .isotp_user_send_can=send_can,
    .isotp_user_get_ms=millis,
    .isotp_user_debug=NULL,
    .user_data=NULL,
};

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud)
{
    //mu own can send api
    return size;
}

//in other task deal canbus msg
bool decode(uint32_t id,uint8_t buf[],uint8_t len)
{
    if ( len == 0 )
    {
        return false;
    }

    if (len > 8)
    {
        log_d("CAN packet too long, truncating");
        len = 8;
    }

    if (id == tp.phys_sa)
    {
        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("phys frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.phys_link, buf, len);
        return true;
    }

    if (id == tp.func_sa)
    {
        if (ISOTP_RECEIVE_STATUS_IDLE != tp.phys_link.receive_status)
        {
            log_d("func frame received but cannot process because link is not idle");
            return  true;
        }

        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("func frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.func_link, buf, len);
        return true;
    }

    return false;
}

uint8_t fn(UDSServer_t *srv, UDSServerEvent_t ev, const void *arg)
{
    switch (ev)
    {
    /* 0x10服务 */
    case UDS_SRV_EVT_DiagSessCtrl:
    {
        UDSDiagSessCtrlArgs_t *r = (UDSDiagSessCtrlArgs_t *)arg;
        switch (r->type)
        {
        case kSafetySystemDiagnostic:
        case kDefaultSession:
            return kPositiveResponse;
        case kProgrammingSession:
        case kExtendedDiagnostic:
            if (srv->securityLevel > 0)
            {
                return kPositiveResponse;
            }

            return kSecurityAccessDenied;
        default:
            return kSubFunctionNotSupported;
        }
    }

    /* 0x27服务 */
    case UDS_SRV_EVT_SecAccessRequestSeed:
    {
        const uint8_t seed[] = {0x11, 0x22, 0x33, 0x44};
        UDSSecAccessRequestSeedArgs_t *r = (UDSSecAccessRequestSeedArgs_t *)arg;
        return r->copySeed(srv, seed, sizeof(seed));
    }

    /* 0x28服务 */
    case UDS_SRV_EVT_CommCtrl:
        return kPositiveResponse;

    /* 会话溢出 */
    case UDS_SRV_EVT_SessionTimeout:
        log_d("server session timed out!\n");
        init();
        return kPositiveResponse;

    case UDS_SRV_EVT_DoScheduledReset:
        log_d("powering down!\n");
        return kPositiveResponse;
    default:
        log_d("Unhandled event: %d\n", ev);
        return kServiceNotSupported;
    }
}

void init()
{
    assert(UDS_OK == UDSServerInit(&srv));
    assert(UDS_OK == UDSISOTpCInit(&tp, &tp_cfg));

    srv.tp = &tp.hdl;
    srv.fn = fn;
}

void loop()
{
    UDSServerPoll(&srv);
}

}

matlam1989, 2024-08-19:
I use C++ for development, and there are very few changes to the library files.more detail:
`#include "elog.h"
extern "C"{
#include "server.h"
#include "isotp_c.h"
}

namespace udsCfg{

extern "C" uint32_t millis();

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud);

UDSServer_t srv;
UDSISOTpC_t tp;

const UDSISOTpCConfig_t tp_cfg =
{
    .source_addr=0x7C0,             
    .target_addr=0x7C8,            
    .source_addr_func=0x7DF,        
    .target_addr_func=UDS_TP_NOOP_ADDR,
    .isotp_user_send_can=send_can,
    .isotp_user_get_ms=millis,
    .isotp_user_debug=NULL,
    .user_data=NULL,
};

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud)
{
    //mu own can send api
    return size;
}

//in other task deal canbus msg
bool decode(uint32_t id,uint8_t buf[],uint8_t len)
{
    if ( len == 0 )
    {
        return false;
    }

    if (len > 8)
    {
        log_d("CAN packet too long, truncating");
        len = 8;
    }

    if (id == tp.phys_sa)
    {
        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("phys frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.phys_link, buf, len);
        return true;
    }

    if (id == tp.func_sa)
    {
        if (ISOTP_RECEIVE_STATUS_IDLE != tp.phys_link.receive_status)
        {
            log_d("func frame received but cannot process because link is not idle");
            return  true;
        }

        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("func frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.func_link, buf, len);
        return true;
    }

    return false;
}

uint8_t fn(UDSServer_t *srv, UDSServerEvent_t ev, const void *arg)
{
    switch (ev)
    {
    /* 0x10服务 */
    case UDS_SRV_EVT_DiagSessCtrl:
    {
        UDSDiagSessCtrlArgs_t *r = (UDSDiagSessCtrlArgs_t *)arg;
        switch (r->type)
        {
        case kSafetySystemDiagnostic:
        case kDefaultSession:
            return kPositiveResponse;
        case kProgrammingSession:
        case kExtendedDiagnostic:
            if (srv->securityLevel > 0)
            {
                return kPositiveResponse;
            }

            return kSecurityAccessDenied;
        default:
            return kSubFunctionNotSupported;
        }
    }

    /* 0x27服务 */
    case UDS_SRV_EVT_SecAccessRequestSeed:
    {
        const uint8_t seed[] = {0x11, 0x22, 0x33, 0x44};
        UDSSecAccessRequestSeedArgs_t *r = (UDSSecAccessRequestSeedArgs_t *)arg;
        return r->copySeed(srv, seed, sizeof(seed));
    }

    /* 0x28服务 */
    case UDS_SRV_EVT_CommCtrl:
        return kPositiveResponse;

    /* 会话溢出 */
    case UDS_SRV_EVT_SessionTimeout:
        log_d("server session timed out!\n");
        init();
        return kPositiveResponse;

    case UDS_SRV_EVT_DoScheduledReset:
        log_d("powering down!\n");
        return kPositiveResponse;
    default:
        log_d("Unhandled event: %d\n", ev);
        return kServiceNotSupported;
    }
}

void init()
{
    assert(UDS_OK == UDSServerInit(&srv));
    assert(UDS_OK == UDSISOTpCInit(&tp, &tp_cfg));

    srv.tp = &tp.hdl;
    srv.fn = fn;
}

void loop()
{
    UDSServerPoll(&srv);
}

}

`

matlam1989, 2024-08-19:
``
#include "elog.h"
extern "C"{
#include "server.h"
#include "isotp_c.h"
}

namespace udsCfg{

extern "C" uint32_t millis();

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud);

UDSServer_t srv;
UDSISOTpC_t tp;

const UDSISOTpCConfig_t tp_cfg =
{
    .source_addr=0x7C0,             
    .target_addr=0x7C8,            
    .source_addr_func=0x7DF,        
    .target_addr_func=UDS_TP_NOOP_ADDR,
    .isotp_user_send_can=send_can,
    .isotp_user_get_ms=millis,
    .isotp_user_debug=NULL,
    .user_data=NULL,
};

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud)
{
    //mu own can send api
    return size;
}

//in other task deal canbus msg
bool decode(uint32_t id,uint8_t buf[],uint8_t len)
{
    if ( len == 0 )
    {
        return false;
    }

    if (len > 8)
    {
        log_d("CAN packet too long, truncating");
        len = 8;
    }

    if (id == tp.phys_sa)
    {
        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("phys frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.phys_link, buf, len);
        return true;
    }

    if (id == tp.func_sa)
    {
        if (ISOTP_RECEIVE_STATUS_IDLE != tp.phys_link.receive_status)
        {
            log_d("func frame received but cannot process because link is not idle");
            return  true;
        }

        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("func frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.func_link, buf, len);
        return true;
    }

    return false;
}

uint8_t fn(UDSServer_t *srv, UDSServerEvent_t ev, const void *arg)
{
    switch (ev)
    {
    /* 0x10服务 */
    case UDS_SRV_EVT_DiagSessCtrl:
    {
        UDSDiagSessCtrlArgs_t *r = (UDSDiagSessCtrlArgs_t *)arg;
        switch (r->type)
        {
        case kSafetySystemDiagnostic:
        case kDefaultSession:
            return kPositiveResponse;
        case kProgrammingSession:
        case kExtendedDiagnostic:
            if (srv->securityLevel > 0)
            {
                return kPositiveResponse;
            }

            return kSecurityAccessDenied;
        default:
            return kSubFunctionNotSupported;
        }
    }

    /* 0x27服务 */
    case UDS_SRV_EVT_SecAccessRequestSeed:
    {
        const uint8_t seed[] = {0x11, 0x22, 0x33, 0x44};
        UDSSecAccessRequestSeedArgs_t *r = (UDSSecAccessRequestSeedArgs_t *)arg;
        return r->copySeed(srv, seed, sizeof(seed));
    }

    /* 0x28服务 */
    case UDS_SRV_EVT_CommCtrl:
        return kPositiveResponse;

    /* 会话溢出 */
    case UDS_SRV_EVT_SessionTimeout:
        log_d("server session timed out!\n");
        init();
        return kPositiveResponse;

    case UDS_SRV_EVT_DoScheduledReset:
        log_d("powering down!\n");
        return kPositiveResponse;
    default:
        log_d("Unhandled event: %d\n", ev);
        return kServiceNotSupported;
    }
}

void init()
{
    assert(UDS_OK == UDSServerInit(&srv));
    assert(UDS_OK == UDSISOTpCInit(&tp, &tp_cfg));

    srv.tp = &tp.hdl;
    srv.fn = fn;
}

void loop()
{
    UDSServerPoll(&srv);
}

}

``

matlam1989, 2024-08-19:
I use C++for development, and there are very few changes to the library files.  IDE uses macro definitions such as：UDS_TP_ISOTP_C,ISOTP_BYTE_ORDER_LITTLE_ENDIAN.
more detail:

#include "elog.h"
extern "C"{
#include "server.h"
#include "isotp_c.h"
}

namespace udsCfg{

extern "C" uint32_t millis();

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud);

UDSServer_t srv;
UDSISOTpC_t tp;

const UDSISOTpCConfig_t tp_cfg =
{
    .source_addr=0x7C0,             
    .target_addr=0x7C8,            
    .source_addr_func=0x7DF,        
    .target_addr_func=UDS_TP_NOOP_ADDR,
    .isotp_user_send_can=send_can,
    .isotp_user_get_ms=millis,
    .isotp_user_debug=NULL,
    .user_data=NULL,
};

int send_can(const uint32_t arb_id, const uint8_t *data, const uint8_t size, void *ud)
{
    //mu own can send api
    return size;
}

//in other task deal canbus msg
bool decode(uint32_t id,uint8_t buf[],uint8_t len)
{
    if ( len == 0 )
    {
        return false;
    }

    if (len > 8)
    {
        log_d("CAN packet too long, truncating");
        len = 8;
    }

    if (id == tp.phys_sa)
    {
        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("phys frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.phys_link, buf, len);
        return true;
    }

    if (id == tp.func_sa)
    {
        if (ISOTP_RECEIVE_STATUS_IDLE != tp.phys_link.receive_status)
        {
            log_d("func frame received but cannot process because link is not idle");
            return  true;
        }

        #if ELOG_OUTPUT_LVL >= ELOG_LVL_DEBUG
        elog_hexdump_msg("func frame Rx", 8, buf, len);
        #endif
        isotp_on_can_message(&tp.func_link, buf, len);
        return true;
    }

    return false;
}

uint8_t fn(UDSServer_t *srv, UDSServerEvent_t ev, const void *arg)
{
    switch (ev)
    {
    case UDS_SRV_EVT_DiagSessCtrl:
    {
        UDSDiagSessCtrlArgs_t *r = (UDSDiagSessCtrlArgs_t *)arg;
        switch (r->type)
        {
        case kSafetySystemDiagnostic:
        case kDefaultSession:
            return kPositiveResponse;
        case kProgrammingSession:
        case kExtendedDiagnostic:
            if (srv->securityLevel > 0)
            {
                return kPositiveResponse;
            }

            return kSecurityAccessDenied;
        default:
            return kSubFunctionNotSupported;
        }
    }

    case UDS_SRV_EVT_SecAccessRequestSeed:
    {
        const uint8_t seed[] = {0x11, 0x22, 0x33, 0x44};
        UDSSecAccessRequestSeedArgs_t *r = (UDSSecAccessRequestSeedArgs_t *)arg;
        return r->copySeed(srv, seed, sizeof(seed));
    }
    case UDS_SRV_EVT_CommCtrl:
        return kPositiveResponse;

    case UDS_SRV_EVT_SessionTimeout:
        log_d("server session timed out!\n");
        init();
        return kPositiveResponse;

    case UDS_SRV_EVT_DoScheduledReset:
        log_d("powering down!\n");
        return kPositiveResponse;
    default:
        log_d("Unhandled event: %d\n", ev);
        return kServiceNotSupported;
    }
}

void init()
{
    assert(UDS_OK == UDSServerInit(&srv));
    assert(UDS_OK == UDSISOTpCInit(&tp, &tp_cfg));

    srv.tp = &tp.hdl;
    srv.fn = fn;
}

void loop()
{
    UDSServerPoll(&srv);
}

}

driftregion, 2024-08-19:
Is it possible to write a failing unit test that shows this bug?  This would constitute a [Minimum reproducible example](https://en.wikipedia.org/wiki/Minimal_reproducible_example).  For example, something like this: https://github.com/driftregion/iso14229/blob/main/test/test_server_0x10_diag_sess_ctrl_functional_request.c

When you send me only code snippets without a CAN log I cannot tell exactly where the issue is. I can only guess. This is not a good use of time.

matlam1989, 2024-08-19:
I get it , and test the program on my computer after work. Thanks

closed 2025-03-09 by driftregion