driftregion/iso14229 #45: use iso14229 on windows

momet83, 2024-11-05:
Hi Driftregion,

I'm new on githup. I want to use your project. ich have 2xESP32 they are connected via CAN and I use VS code.
now to my quations:

1. wich folder should I download? Windows or main?
2. next step: I complie the ESP32_server and flash the code on ESP32
3. then I compile for example "test/test_server_0x22_RDBI.c" 
4.  how can I send the request to ESP32?

driftregion, 2024-11-06:
Hi @momet83, 

> 1. wich folder should I download? Windows or main?

use the main branch. Download the source files from the releases link https://github.com/driftregion/iso14229/releases/tag/v0.7.1 , copy `iso14229.c` and `iso14229.h` into your source tree and build.

> 4. how can I send the request to ESP32?

For the client (sending a request to the ESP32), I'd recommend using https://github.com/pylessard/python-udsoncan/

momet83, 2024-11-27:
Hi @driftregion,
Thank you for your support. I have download the source files as you have said. I have used your folder “ESP32_server” but after compiling the VS code shows me a problem “unused variable ‘ta’” is it a problem or I can ignore it and I'll start flash the esp32? 
[image]

2nd Question: in the folder "iso14229-0.7.1 --> test" dor what are the tests there?

thanks in advance!

closed 2025-03-09 by driftregion