driftregion/iso14229 #47: UDS_VERSION not handled correctly/automatically

muehlke, 2024-11-13:
@driftregion I was just wondering how the versioning mechanism has been decided for this repo. The `src/version.h` points at `0.7.0` and the latest release at `v0.7.1`. On the other side, I don't see some logic for incrementing the versions on the `.github/workflows/release.yml`. Is this maybe planned for the future? Otherwise, before every official release one would have to add the new version to `src/version.h` before merging, which can be easily forgotten.

Best regards :v:

driftregion, 2024-11-13:
Great points, thanks! I would like to do a release checklist similar to https://sqlite.org/src/ext/checklist/3270000/index , but haven't gotten around to it yet. Here's a rough draft:

Checklist:
- [ ] increment version in `src/version.h`
- [ ] `git tag` with release version (the current implementation of `.github/workflows/release.yml` grabs the release triple from the git tag)

driftregion, 2025-01-24:
Release checklist updated in README.md

closed 2025-01-24 by driftregion

muehlke, 2025-01-24:
Great job! 💯