driftregion/iso14229 #57: Security Access state following a forced transition to the default session

lee-matthews-decathlon, 2025-03-03:
Hi

I'm trying to understand how to handle security access levels following a timeout due to no testerpresent messages.

Currently, if I change session to user, send a requestseed message followed by a sendkey message, I can then access any protected sessions that I've implemented.

However, if the testerpresent messages stop, my state machine transitions back into the defaultsession, but the security access remains. If I perform another requestseed, it returns unlocked, which is as expected given the code.

The ISO14229-1 specification (page 38), states that when we transition back to the default session, "a locked security level shall be activated in the server". How can I inform the iso14229 library that this transition to the default session has taken place and it should return to a locked state? If I understand correctly, this isn't currently implemented ?

I've added a workaround by adding a function such as 

void UDSServerSecurityAccessReset(UDSServer_t *srv)
{
   srv->securityLevel = 0;
}

Any input would be appreciated.

Thanks in advance
Lee

driftregion, 2025-03-03:
Hi Lee,

Thank you for this detailed issue. I think you're correct that this
behavior is not correctly implemented.

My read is that we should solve this within the library by identifying
tester present timeout and reverting the security level to zero
automatically. Would this solve your use case?

If you could provide some failing test cases in test_server.c, that would
be great.

Kind regards,

Nick

lee-matthews-decathlon, 2025-03-04:
Thanks for getting back to me Nick.

Sure, I'll have a look at writing some failing test cases. Currently I'm struggling to get the tests to run : 

bazel test //test:all
Starting local Bazel server and connecting to it...
ERROR: /home/lee/Dev/can/iso14229/test/BUILD:52:12: //test:test_client_0x11_ECU_reset_tp_mock: no such attribute 'env' in 'cc_test' rule
ERROR: /home/lee/Dev/can/iso14229/test/BUILD:52:12: //test:test_client_0x22_RDBI_unpack_response_tp_mock: no such attribute 'env' in 'cc_test' rule
ERROR: /home/lee/Dev/can/iso14229/test/BUILD:52:12: //test:test_client_0x31_RCRRP_tp_mock: no such attribute 'env' in 'cc_test' rule

I cannot find any instructions that explain what I need to do with the env attribute. Could you help please?

Lee

driftregion, 2025-03-04:
What is your bazel version?

lee-matthews-decathlon, 2025-03-04:
bazel 3.5.1- (@non-git)

lee-matthews-decathlon, 2025-03-13:
Hi

I had some problems getting the tests working.

Firstly, I had a lot of certificate problems, though this may be due to my corporate firewall.

This was fixed by getting the .certificate file from the https://bcr.bazel.build website, then adding it to the java keystore.

I had to run the command bazel --host_jvm_args="-Djavax.net.ssl.trustStore=/etc/ssl/certs/java/cacerts" test //test:all

to resolve the certificate problem. However, this produced another issue : 

bazel --host_jvm_args="-Djavax.net.ssl.trustStore=/etc/ssl/certs/java/cacerts" test //test:all
WARNING: Running Bazel server needs to be killed, because the startup options are different.
Starting local Bazel server (8.1.1) and connecting to it...
ERROR: no such package '@@[unknown repo 'platforms' requested from @@]//cpu': The repository '@@[unknown repo 'platforms' requested from @@]' could not be resolved: No repository visible as '@platforms' from main repository. Was the repository introduced in WORKSPACE? The WORKSPACE file is disabled by default in Bazel 8 (late 2024) and will be removed in Bazel 9 (late 2025), please migrate to Bzlmod. See https://bazel.build/external/migration
ERROR: Analysis of target '//test:test_prefix' failed; build aborted: no such package '@@[unknown repo 'platforms' requested from @@]//cpu': The repository '@@[unknown repo 'platforms' requested from @@]' could not be resolved: No repository visible as '@platforms' from main repository. Was the repository introduced in WORKSPACE? The WORKSPACE file is disabled by default in Bazel 8 (late 2024) and will be removed in Bazel 9 (late 2025), please migrate to Bzlmod. See https://bazel.build/external/migration
INFO: Elapsed time: 20.488s, Critical Path: 0.03s
INFO: 1 process: 1 internal.
ERROR: Build did NOT complete successfully
ERROR: No test targets were found, yet testing was requested

I solved this by moving from bazel 8.1.1 to bazel 7.5.0.

There are a lot of tests that fail - is this normal? I've attached a log of my tests in the main branch.

[bazel-test-resutls.txt](https://github.com/user-attachments/files/19229583/bazel-test-resutls.txt)

driftregion, 2025-03-23:
Hi @lee-matthews-decathlon , I had a look at the log.  The failing tests are likely due to lack of `vcan` virtual socketcan interface and are expected unless you've configured one on your machine.

I see that you are using an older version of the library. The test suite was refactored in #51 

Here's an example test of `0x10` that you might use as a starting point:
https://github.com/driftregion/iso14229/blob/main/test/test_server.c#L102

lee-matthews-decathlon, 2025-03-31:
The vcan fixed it, thanks. I've equally updated to the latest version.

I've started working on a unit test, but I keep getting an assertion and I'm not sure why.

Here is the code : https://pastebin.com/vdWSbjUA

I tried pushing a branch but I don't have write access to your repo.

V (37656) src/tp/isotp_mock.c: TPMock: detached client. TPCount: 0
[       OK ] test_0x27_brute_force_prevention_2
[ RUN      ] test_0x27_security_access_loss_after_timeout
test_server_tp_mock: test/test_server.c:316: fn_test_0x27_security_access: Assertion `0' failed.

The assertion seems to be linked to the timedelay here : 

 // Wait for timeout - simulate no tester present messages
    EnvRunMillis(e, UDS_SERVER_DEFAULT_S3_MS + 10);

If I reduce the delay to about 5047, I no longer get the assert. I don't know why. Maybe I'm missing something?

driftregion, 2025-04-09:
> Here is the code:

Please use backticks to embed code as follows:
<pre>
```c
void code();
```
</pre>

which renders like this: 
```c
void code();
```

> I tried pushing a branch but I don't have write access to your repo.

Use the fork button to make a public copy of the repo which you may push to. You can later use this to open a pull request.
[image]

When attaching output like this, it is ambiguous which line of code `316` is unless you have also posted a copy of the code. Please make a fork, push your code, and link to it.
```
test_server_tp_mock: test/test_server.c:316: fn_test_0x27_security_access: Assertion `0' failed.
```

You can link to specific code snippets on your fork like this:

https://github.com/driftregion/iso14229/blob/4a9394163d817b6ec55c3ad5c8e937bd51efeadf/test/test_server.c#L121-L122

> The assertion seems to be linked to the timedelay here :

The current implementation sends a `UDS_EVT_SessionTimeout` when the session timeout has elapsed.

https://github.com/driftregion/iso14229/blob/4a9394163d817b6ec55c3ad5c8e937bd51efeadf/src/server.c#L965

This happens at approximately `UDS_SERVER_DEFAULT_S3_MS` after the most recent received message.
 https://github.com/driftregion/iso14229/blob/4a9394163d817b6ec55c3ad5c8e937bd51efeadf/src/config.h#L54

Rereading your original post https://github.com/driftregion/iso14229/issues/57#issue-2891748269 , my understanding is that the meaning of different security levels is **user-defined**.

Would it work to handle the `UDS_EVT_SessionTimeout` as follows?

```c
int fn(UDSServer_t *srv, UDSEvent_t ev, void *arg) {
    switch(ev) {
        case UDS_EVT_SessionTimeout:
            srv->securityLevel = 0;
            return UDS_OK;
    }
}
```