driftregion/iso14229 #64: Diagnostic Session Control type is masked with 0x4F

NicolasCayoux, 2025-03-25:
Hi,

On UDS Server, function "uint8_t _0x10_DiagnosticSessionControl(UDSServer_t *srv, UDSReq_t *r)"

The session type is masked with 0x4F
"uint8_t sessType = r->recv_buf[1] & 0x4F;"

Is there a specific reason to use this mask, instead of 0x7F ?
According to several UDS example on internet, the session type can be set between 0x01 and 0x7F

driftregion, 2025-03-26:
Hi @NicolasCayoux , you're right, it should be 0x7F. Please feel free to open a PR.

NicolasCayoux, 2025-03-27:
Hi @driftregion,
I try to open a branch "issue_64" to edit the code, but I don't have the permission to push the code.

The modification is
file src/server.c
ligne 34 (function static uint8_t _0x10_DiagnosticSessionControl(UDSServer_t *srv, UDSReq_t *r) )
    uint8_t sessType = r->recv_buf[1] & 0x4F;
=>
    uint8_t sessType = r->recv_buf[1] & 0x7F;

commit 3c3c9585 (2025-08-15): Merge pull request #69 from driftregion/fix_64

fix #64
commit 5d77d763 (2025-08-15): fix #64